mangdangroboticsclub/mini_pupper #4: How to begin developing this Open Source Project. (Software, Custom Parts)

RedgarOpenSource, 2021-08-24:
@mangdangroboticsclub Hello! I am very interested in building this project. I love this project so much! There are just some things I am confused about.

1: I understand that there is a BOM list but I don't know where to order the custom Carbon fiber parts. I will order from the section that says Custom parts but I don't see the Carbon fiber shell there. If I can print the Carbon Fiber shell using standard PLA, please let me know. 

2: I also noticed that the Software Installation tutorial is missing. If possible, can you provide a link?

I will also try to add new features to this bot such as GPS location and more. I would then update this Open Source Project with access to its code. 

Thank you so much in advance!

mangdangroboticsclub, 2021-08-24:
Hey Redgar,
Thanks for reaching out.
1. You can find the 3D file [here](https://a360.co/3fS15a1), actually, I prefer Carbon Fiber but not PLA.
2. Please refer to the below document.
https://github.com/mangdangroboticsclub/QuadrupedRobot/blob/MiniPupper/Doc/guide/software_installation.rst

Very appreciated if you can improve Mini Pupper.
Before start, could you please send me a mail ( afreez@mangdang.net ) and let me know your background and schedule?
I will also sync our status and roadmap with you.
Best Regards,
Afreez

RedgarOpenSource, 2021-08-24:
Thank you so much for your comment, I just emailed you.