mangdangroboticsclub/mini_pupper #7: Is there a Gazebo simulation model for mini-pupper?

mkhansenbot, 2021-10-04:
Is there a Gazebo simulation model (URDF) for mini-pupper? If not, are the CAD files available to create one?

mangdangroboticsclub, 2021-10-05:
Will try to update the files before this weekend, thanks!

mkhansenbot, 2021-10-15:
Any update on a simulation model (URDF)?

hanxiaotian, 2021-11-08:
also interested, thanks!

mhered, 2022-01-12:
Check out this article (original in japanese I am linking the google translate version: [How to move Mini Pupper (Gazebo)](https://qiita-com.translate.goog/nisshan_/items/fa4a4ab807c5e4bb1ff4?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=es)

mangdangroboticsclub, 2022-01-28:
Preparing the Simulation documents will release it soon, thanks!

mangdangroboticsclub, 2022-02-17:
Please confirm the below link, thanks!
https://github.com/mangdangroboticsclub/aws-robomaker-mini-pupper-application

closed 2022-02-17 by mangdangroboticsclub

BarrettBytes, 2024-03-22:
the model runs unreasonably slowly on my computer, is there any reason for this? gazebo setting i need to change or?