mangdangroboticsclub/mini_pupper #16: Lidar installation:

EdSeymore, 2022-01-21:
Do you have instructions for mechanical installation of the provided lidar?

EdSeymore, 2022-01-21:
Sorry, this question was meant for the Mini Pupper. 

Do you have a link to support for the Min Pupper?

tractrac91, 2022-01-24:
If you have the "DTOF LiDAR LD06" sensor, you can find the manuals at the website from the manufactorer: https://www.ldrobot.com/download/en/98

Important if you follow the instructions in the development manual: At Ubuntu 18.04 you **can't** install the kinetic version of ROS. I used the melodic version of ROS and it worked fine for me.

mangdangroboticsclub, 2022-01-24:
For the document, please refer to the below link at first, and will add the Lidar section guide very soon this week.
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/index.html
Best,

mangdangroboticsclub, 2022-02-09:
Please confirm the below link for Lidar part, thanks!
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/guide/SLAM.html

closed 2022-02-09 by mangdangroboticsclub