mangdangroboticsclub/mini_pupper #18: Assembly Instructions instead of placeholders?

axello, 2022-01-25:
Hi,
It seems most assembly instructions, apart from the Legs video are simply placeholders for missing, euh, _Assembly Instructions_

Do they still need to be copied from another repository, or are these documented elsewhere?

Blackvenom, 2022-01-26:
Workaround:
If you Switch on top of minipupper GitHub Page from MiniPupper_V2 to MiniPupper_V1, you can see assembly instructions on the bottom:
https://drive.google.com/file/d/12Xbnqv3Rm7TEcjBZVSbVkBVNKNO3IOfm/view?usp=sharing

tractrac91, 2022-01-26:
You can find newer instructions here: [https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/guide/Assembly.html](https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/guide/Assembly.html)

mangdangroboticsclub, 2022-01-28:
Yes, about the documents, please follow the below link,
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/index.html

closed 2022-01-28 by mangdangroboticsclub