mangdangroboticsclub/mini_pupper #20: Battery monitor not working after upgrading ubuntu 21.04

jensanjo, 2022-01-27:
I used the [20220109_v1.7.0_MiniPupper_V2_PS4_Ubuntu_21.10.0_Baseline.img.zip](https://drive.google.com/file/d/1ZYJCfCb-1U71M0wobJEzo1sL0uNgFjqz/view?usp=sharing) image.
The battery monitor was working. I could read the data from `/sys/class/power_supply/max1720x_battery`.
After installing I did a `apt upgrade` the battery monitor was no longer working.
During the upgrade a new kernel was installed (`5.13.0-1015-raspi`). The previous kernel was `5.13.0-1011-raspi`.
It turns out that the `max1720x_battery` kernel module is not present in the new kernel.

tech-life-hacking, 2022-02-07:
I also got same thing after apt upgrade, but 
the battery monitor recovered by running the script below in my case.
https://github.com/mangdangroboticsclub/QuadrupedRobot/blob/MiniPupper_V2/Legacy/pre_install.sh

tech-life-hacking, 2022-02-09:
Sorry, what I mention above, only for not apt upgrade.
In my case, I installed some ROS packages and the battery monitor didn't work,
but it recovered correctly after running scripts above.

mangdangroboticsclub, 2022-02-14:
@jensanjo  Yes , you are right . we will open the driver source and share the way to rebuild them.  then , this feature can be solved. thanks for your feedback.

mangdangroboticsclub, 2022-02-14:
@tech-life-hacking  this issue is just on the device who upgrade the kernel .