mangdangroboticsclub/mini_pupper #21: calibrate_tool not able to access calibration file

jensanjo, 2022-01-27:
I have issues with the calibrate tool. At start it gives an error message:

`Error, get nv calibration params failed, use default value. Please calibrate your pupper !`

When I press the 'Update' button I get this error:
```python
out
[[ -9.   9.  12.  15.]
 [ 35.  35.  60.  35.]
 [-30. -27. -22. -48.]]
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "calibrate_tool.py", line 239, in updateButton2Event
    self.writeCalibrationFile()
  File "calibrate_tool.py", line 193, in writeCalibrationFile
    with open(ServoCalibrationFilePath, "w") as nv_f:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/nvmem/devices/3-00501/nvmem'
``` 
The mentioned path does not exist. Apparently the name is wrong:

```bash
# ls  /sys/bus/nvmem/devices/
3-00500
```
So I changed the path to `/sys/bus/nvmem/devices/3-00500/nvmem` and tried again.
Now it works and there are no error messages

mangdangroboticsclub, 2022-02-09:
I have not found the issue, could you please let me know your image file name? Thanks!

jensanjo, 2022-02-09:
I used `20211220_v2.1.3_MiniPupper_V2_PS4_Ubuntu_20.04.2_Baseline.img`

mangdangroboticsclub, 2022-02-09:
Your version is a little bit old, last year version:)
Please try the 2022 version, such as, 20220109_v1.7.0_MiniPupper_V2_PS4_Ubuntu_21.10.0_Baseline.img.zip , thanks!
https://drive.google.com/drive/folders/17XOR7FHEMkts_zZtZtuQ-QxEJ68U8ZkW

closed 2022-02-09 by mangdangroboticsclub

gilaramirez, 2022-02-14:
I just ran into the same issue with the 2022 version 2 file posted as the solution above.  This bug is not fixed.