mangdangroboticsclub/mini_pupper #26: is there an updated build instruction

lein1013, 2022-02-06:
comparing the parts I got with the BOM I see that there are differences in length and size - e.g at the inserts and the standoffs
Is there an update build instruction available?

axello, 2022-02-07:
I would like one as well! 
Even though I succeeded in putting the mini pupper together, I seem to be missing lots of 4 mm screws for the leg covers. 

I did discovered a newer manual online in Chinese, *with* all the necessary screw lengths. Their google drive is more up to date.
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/guide/Assembly.html

hubertsmits, 2022-02-13:
My apologies if I’m not logging this request correctly. I broke the LCD for the mini puppet (the 3D prints are pretty bad). How can I buy a replacement LCD?

— Hubert

	Message ID: ***@***.***>

gilaramirez, 2022-02-14:
Can you please remove the instructions from this site's wiki page, as they are not correct. I spent hours trying to flash firmware unsuccessfully before I found the alternative website with the correct files. Please replace with a link to the maintained site.

mangdangroboticsclub, 2022-03-02:
Sorry for the delay, already update the site's wiki page.
Please follow the document at the below link, thanks!
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/index.html

closed 2022-03-02 by mangdangroboticsclub