mangdangroboticsclub/mini_pupper #33: Google drive link for Assembly instructions doesn't work

thaytan, 2022-02-24:
The link given for Assembly instructions in the README (https://drive.google.com/file/d/12Xbnqv3Rm7TEcjBZVSbVkBVNKNO3IOfm/view?usp=sharing) leads me to a 'Sorry, the file you have requested does not exist.' error.

[image]

TheFoolishPupil, 2022-02-28:
There are assembly instructions in the docs [here](https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/guide/Assembly.html).

mangdangroboticsclub, 2022-03-02:
Thanks, TheFoolishPupil.
Hi thaytan, please refer to the document at the below link, thanks!
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/index.html

closed 2022-03-02 by mangdangroboticsclub