mangdangroboticsclub/mini_pupper #34: Missing Parts / Files / URDF / Specifications

Danfoa, 2022-02-25:
Good day, 

I have not received several of the parts advertised with the developer package, but what it is more concerning there is no information about these parts anywhere, making it impossible to buy them, or print them. 

Regarding missing parts information in the BOM Lists:
-  The 4 bearings/brushings used in the chassis of the robot are not listed, and its specifications are not known making it impossible to buy these parts (WHICH DID NOT COME WITH MY PACKAGE). Are these bearings? or brushings? please provide a link to them or at least their specifications.
[image]

Regarding missing STL files:
- Robot feet. I did not receive the robot feet with my package, there is not 3D file to them, nor a mechanism for ordering them

**Can you share the Fusion360 assembly file, such that we can use it for more than visualization, that is, so we can use this assembly to obtain a URDF file and to properly make modifications to the robot parts, and assembly?.**

TheFoolishPupil, 2022-02-28:
Hey @Danfoa, you can find the STL files [here](https://drive.google.com/drive/folders/1qY0D0EFMEq7hCxesOtWjdjGncZ6CRQnF). The link is from this page in the docs - https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/reference/Design.html

mangdangroboticsclub, 2022-03-02:
Hi Danfoa,

Maybe you ignored our Kickstarter update, please refer to the below link, 
https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/index.html

You can also find the BOM list on the Kickstarter page and the detailed components information at the below link,
https://drive.google.com/file/d/18phJat8GdK5Yq5p4K1ZmfY1-nMf1lQw4/view

Thanks!

closed 2022-03-02 by mangdangroboticsclub