mangdangroboticsclub/mini_pupper #37: Issues with leg 4 thigh

TheFoolishPupil, 2022-02-28:
Hey there. I'm seeing issues with the movement for what seems to be the thigh motor on leg 4. You can see from the video that it seems to miss the full range of movement. In the calibration tool it has no problems with the full range.

https://user-images.githubusercontent.com/43202406/156047351-da76b2b4-ed9e-4ce8-a07e-ce17d6f27959.mov

gilaramirez, 2022-03-02:
I am having a similar problem with leg 1 calf.  It began after I attempted to install the SLAM software. I have tried the following troubleshooting steps:
1. Recalibration - the motor moves to the calibrated position.  When I turn on the calibration application. It will not respond to any adjustments to the calibration slider. 
2. I flashed the OS again.
3. I swapped the motor with a spare.  
Nothing has resulted in fixing the problem. I took a video of the issue.
[Video](https://photos.app.goo.gl/wyLC5u2fnDqBkCiw8)

gilaramirez, 2022-03-04:
It does not appear the issue is about calibration itself, unless the Calibration application is somehow setting a value in the matrix that is out of bounds for the software, thus causing the custom board to stop sending signals to the affected motor.

TannerGilbert, 2022-03-06:
I have a similar issue as @TheFoolishPupil. I tried switching the servo with the substitute servo that was provided with the kit but that also didn't fix the issue.

feiticeir0, 2022-03-06:
I'm also having an issue with calf of leg 4. It just doesn't move.  All the other ones are fine, but on the calibration tool, the slider doesn't move the calf... Will try to replace the motor with the spare one and see.

decade549, 2022-03-07:
By my Pupper the Problem is in the servo 5, which is pin 11 on the raspberry.
So you need to change the pin of the servo .By your Pupper leg, I would say it is the servo 8 which is pin 8 on the raspberry pi. change it to any remainding from 3 free pin left (it counting start from 1 from the out side). 
When you done with the changing the Pin. Turn on your raspberry find the file Config.py in the Standford\Pupper folder. Open the Config file and change the pin number accordingly to your pin on raspberry. 
Hope this will help you like in and my Case.

TannerGilbert, 2022-03-07:
@decade549 Is there some wiring diagram that shows which servo connector is connected to which pin on the Raspberry Pi?

decade549, 2022-03-08:
> @decade549 Is there some wiring diagram that shows which servo connector is connected to which pin on the Raspberry Pi?

Hope this help. I using the ubuntu version 21.10 .So the config file is difference
[Pin matrix.xlsx](https://github.com/mangdangroboticsclub/QuadrupedRobot/files/8204348/Pin.matrix.xlsx)

gilaramirez, 2022-03-08:
> @decade549 Is there some wiring diagram that shows which servo connector is connected to which pin on the Raspberry Pi?

Look at the image in the assembly page.  Link here:

https://mangdang-minipupperdocs.readthedocs-hosted.com/en/latest/_images/52.png

TannerGilbert, 2022-03-09:
@decade549 thanks for the useful document. In my case it's servo 6 that is not working correctly. Therefore I rewired servo 5 to pin 3 (from pin 10) and changed the pin in the `PWMParams`.

```
self.pins = np.array([[15, 12, 9, 6], [14, 11, 8, 5], [13, 3, 7, 4]])
```

Unfortunately the servo doesn't react when running the calibration tool. I also tried running the calibration tool with Python but it also didn't work:

```
ubuntu@ubuntu-desktop:~/Robotics/QuadrupedRobot/StanfordQuadruped$ sudo python3 calibrate_tool.py
```

Any help is greatly appreciated.

feiticeir0, 2022-03-09:
I have a similar issue, but not with the servo, but with the PIN on the board. 
PIN 11 is not working.. I've switch the servos and the calf that was not working, is working and the thigh that was working, is not working. This takes me to believe that is the PIN on the board. 
Will change to another pin on the config file and see... 

Is my board PIN busted ? 
Thank you

mangdangroboticsclub, 2022-03-09:
Quick Calibration Guide
Please click the "reset" button on the GUI calibration tool and confirm the below link for quick and easy calibration, thanks!
https://minipupperdocs.readthedocs.io/en/latest/guide/Software.html#calibration

If you follow the up guide and still find some servo can not work, please connect the "bad" servo to another pin to check whether the servo is bad or the carry board pin is bad. If there are still other questions, please feel free to let me know, thanks!

feiticeir0, 2022-03-09:
> Quick Calibration Guide Please click the "reset" button on the GUI calibration tool and confirm the below link for quick and easy calibration, thanks! [...]

I don't know if the answer is for me, but I've already changed to another PIN and the servo works. IS THAT PIN THATS BAD... PIN12 is not working... 
Is it possible to change to another PIN ? And why is it not working  ?

gilaramirez, 2022-03-09:
> Quick Calibration Guide Please click the "reset" button on the GUI calibration tool and confirm the below link for quick and easy calibration, thanks! [...]

I have changed the pin and the configuration file and the motor is working.  The carry board is faulty.  I would like to get a replacement carry board. What is the procedure? 

Thanks 

Gil Ramirez 

Thanks,

feiticeir0, 2022-03-09:
> I have changed the pin and the configuration file and the motor is working. The carry board is faulty. I would like to get a replacement carry board. What is the procedure?
>
> Thanks
>
> Gil Ramirez [...]

I have the same problem. Still have to change the PIN in the configuration, but still the same problem.

feiticeir0, 2022-03-09:
> I have changed the pin and the configuration file and the motor is working. The carry board is faulty. I would like to get a replacement carry board. What is the procedure?
>
> Thanks
>
> Gil Ramirez [...]

Where's that configuration file ?

TannerGilbert, 2022-03-09:
> Where's that configuration file ?

`~/Robotics/QuadrupedRobot/StanfordQuadruped/pupper/Config.py`

feiticeir0, 2022-03-09:
> `~/Robotics/QuadrupedRobot/StanfordQuadruped/pupper/Config.py`

Thank you, but still no luck. 
According with the excel from @decade549 ,  Servo 12 (calf) is PIN 4 from Leg4
I've changed in the file from

`  self.pins = np.array([[15, 12, 9, 6], [14, 11, 8, 5], [13, 10, 7, 4]]) `
to
   ` self.pins = np.array([[15, 12, 9, 6], [14, 11, 8, 5], [13, 10, 7, 3]])`

Turned off the RPi
connected the SERVO from J12 to J13 (J12 is not working). 
Turned on the RPi, opened the Calibration tool, tried to move the Calf from LEG4, but nothing... 
I know that the SERVO is working, because if I connect it to another PIN it works...

What am I doing wrong ? 
Thank you

mangdangroboticsclub, 2022-03-10:
Hi guys,
We tested all the carry boards one by one before shipping, but it is difficult to ensure that the board will not be damaged during shipping.
It will be helpful for us to double confirm if you can take a picture and send to  fae@mangdang.net
If we can clearly see the damage through the appearance, we can re-ship it or tell you how to modify it (if you have the tools to fix it).
Best Regards,

feiticeir0, 2022-03-10:
> Hi guys, We tested all the carry boards one by one before shipping, but it is difficult to ensure that the board will not be damaged during shipping. It will be helpful for us to double confirm if you can take a picture and send to [fae@mangdang.net](mailto:fae@mangdang.net) If we can clearly see the damage through the appearance, we can re-ship it or tell you how to modify it (if you have the tools to fix it). Best Regards,

Hi Mangdang !
Thank you for the reply. 

Yesterday I was thinking that this could not be. Following the excellent excel file from @decade549 , I've change the PINs and started the calibration tool and nothing... 

What I hadn't done yet was to try a new servo from the ones extra. That servo did work - not only on the J12 PIN, but also when changing the PIN on the config file. It worked. 

For now It's working , but the other servo it also works... Don't really know what's happening, but, it's fine now. I have movement.

mangdangroboticsclub, 2022-03-10:
Hi feiticeir0, 
Looks great!
Please go ahead!
Thanks!

decade549, 2022-03-11:
> @decade549 thanks for the useful document. In my case it's servo 6 that is not working correctly. Therefore I rewired servo 5 to pin 3 (from pin 10) and changed the pin in the `PWMParams`.
>
> ``` [...]

After changing the config have you reload the robot:
`systemctl stop robot`
`systemctl daemon-reload`
`systemctl reload robot`

mangdangroboticsclub, 2022-03-28:
Hi decade549, 
cc guys,
We tested all the carry boards one by one before shipping, but it is difficult to ensure that the board will not be damaged during shipping.
It will be helpful for us to double confirm if you can take a picture and mail to: fae@mangdang.net 
If we can clearly see the damage through the appearance, we can re-ship it for free or tell you how to modify it (if you have the tools to fix it).
We don't hope you waste your time to debug the carry board, it is not valuable.

Best Regards,

closed 2022-03-28 by mangdangroboticsclub