mangdangroboticsclub/mini_pupper #39: Command is not bind to robot more than a few seconds

feiticeir0, 2022-03-18:
Hi all.

After solving the servo issue, I'm not facing another one. 

My command (the one that came with the KIT) is not bound to the robot more than a few seconds or until a first command. After that, it just turns off (the LED turns off) and the robot does not respond anymore to it. The LCD face also changes to when is not activated. 

I'm following the instructions. The command is bound to the robot - the LED turns green. 
I press the L1 button to activate the robot. It does activate. 
**I press the left joystick - nothing happens.** 
I press the R1 button - from rest to throttle - it starts the throttle - very weirdly I must say. I press the R1 again to rest - 
The green LED turns off... The face of the robot changes... IT does not respond anymore... 

I need to repeat the bind process again... L1 to activate the robot - joystick does nothing... R1 is the only thing that works... 

What's happening ? 

Thank you

mangdangroboticsclub, 2022-03-28:
Looks very strange! 
How about now?
It's hard to give useful advice based on your description.
Could you please let us know the below info at first?  Thanks!
Which image file you are using? or did you build the source code(which branch?) by yourself?

feiticeir0, 2022-03-28:
Hi ! Thank you for the reply. I'm using the latest image available: 20220302_v1.9.0_MiniPupper_V2_PS4_Ubuntu_21.10.0.img.zip

The command is charged - I even left it overnight to make sure. 
I can make a video . Any place that I can see the logs ?

mangdangroboticsclub, 2022-03-28:
Thanks!
Could you please upload your video and share the google drive link?
I will discuss your video with other guys and give you feedback soon.

feiticeir0, 2022-03-28:
Thanks. 
Tomorrow  or Wednesday I'll share the video

mangdangroboticsclub, 2022-04-11:
Let's sync at the below discord channel, maybe it's a better place than here:)
https://discord.gg/enSJfUaV

closed 2022-04-11 by mangdangroboticsclub