mangdangroboticsclub/mini_pupper #40: Running into 'Servos overload please check' issue

gshiva, 2022-04-05:
After many attempts in assembly, I got the mini pupper assembled. After fixing wifi and figuring out how to run calibrate_tool from the command line, I am running into the 'servos overload please check' issue.

https://github.com/mangdangroboticsclub/QuadrupedRobot/blob/32e3b18a2d50e24256dab52251149884d9c1164a/StanfordQuadruped/calibrate_tool.py#L359 

Any suggestions?

mangdangroboticsclub, 2022-04-06:
That means your servo horn position is not right, please refer to the below calibration guide,
https://minipupperdocs.readthedocs.io/en/latest/guide/Software.html#calibration
(remove the 4 legs before calibration)
Thanks!

gshiva, 2022-04-06:
Thanks will try it and will close the issue with instructions on what worked

mangdangroboticsclub, 2022-04-11:
Please join our discord channel for better support, thanks!
https://discord.gg/enSJfUaV

closed 2022-04-11 by mangdangroboticsclub