mangdangroboticsclub/mini_pupper #45: DKMS

aimoff, 2022-06-01:
After updating Ubuntu packages, caribration data was not restored.
I re-instaled initial image, then it worked as expected.

I think it was caused by lack of kernel module for newly installed kernel.
I suggest to use DKMS (Dynamic Kernel Module Support framework) that automatically compiles and installs custom kernel module for newly installed kernel.

mangdangroboticsclub, 2022-06-08:
Many thanks for your feedback.
Is it possible that you or anyone post a pull request to fix it? Actually, I have no background of DKMS.
Looking forward to your help.
Thanks!

hdumcke, 2022-09-24:
We have moved all hardware related stuff to a separate repo (minipupper-bsp) and there is a script to update the kernel modules after reinstalling a new kernel. 

Automating this process with DKMS will be a great benefit, any PR should be made in minipupper-bsp

aimoff, 2022-11-15:
I've not tested the mini_pupper_bsp repository, but I close this issue.