mangdangroboticsclub/mini_pupper #47: Error in SLAM installation guide Section 1.1.1

gilaramirez, 2022-10-29:
The code snippet should read as below (edits in bold text):

cd ~
sudo apt-get update
sudo apt-get install -y python3-wstool python3-rosdep ninja-build stow
mkdir carto_ws
cd carto_ws
wstool init src
wstool merge -t src https://raw.githubusercontent.com/cartographer-project/cartographer_ros/master/cartographer_ros.rosinstall
wstool update -t src
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=**noetic**-y
src/cartographer/scripts/install_abseil.sh
**source /opt/ros/noetic/setup.bash**
sudo apt-get remove ros-${ROS_DISTRO}-abseil-cpp
catkin_make_isolated --install --use-ninja
source install_isolated/setup.bash

Otherwise cartographer_ros will not install.

sskorol, 2022-10-29:
I believe folks intentionally omit sourcing ROS distro as this line should be put into your bash profile. Otherwise, you'll have a hard time sourcing it for every project you have.

gilaramirez, 2022-10-30:
Unfortunately not including the ROS distro at that early stage fails the installation process.  As a matter of fact, following the rest of the SLAM install steps after resolving these still results in a broken installation.  I also found and tried to follow the steps in /src/minippoer_ros/README.md and those steps also failed.

Does anyone have a set of instructions for installation of SLAM?

closed 2022-11-22 by mangdangroboticsclub